favalex/modbus-cli #3: how to install?

eugen80, 2019-06-16:
Hi,

I've installed the script, but after running modbus -h I get:
ImportError: No module named modbus_cli.definitions

modbus_cli should be also installed... what is wrong?

Etantonio2016, 2020-08-10:
same problem for me, how can I solve it?

favalex, 2020-08-15:
What OS are you using? How do you install modbus-cli?

The install instruction in the README work on linux. I don't have access to other OSes to test.

shining-fnml, 2020-11-03:
Hi. I installed the python way (python setup.py install) on Debian 10 Buster x86_64. Python version is 2.7.16 and installation looks fine under usr/local/lib/python2.7/dist-packages/modbus_cli-0.1.4rc2-py2.7.egg
Running modbus I get:
Traceback (most recent call last):
  File "/usr/local/bin/modbus", line 4, in <module>
    __import__('pkg_resources').run_script('modbus-cli==0.1.4rc2', 'modbus')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 666, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1446, in run_script
    exec(code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/modbus_cli-0.1.4rc2-py2.7.egg/EGG-INFO/scripts/modbus", line 8, in <module>
    from modbus_cli.definitions import Definitions
ImportError: No module named modbus_cli.definitions

definitions.py is located in /usr/local/lib/python2.7/dist-packages/modbus_cli-0.1.4rc2-py2.7.egg/modbus_cli and contains class Definitions.

Is there anything wrong with the python include path?

favalex, 2020-12-05:
Please use python 3. Python 2 is not supported.

shining-fnml, 2020-12-06:
> Please use python 3. Python 2 is not supported.

Thanks for replying. Please tell about this to Debian's package mantainer

dragoshenron, 2021-01-06:
Ciao @favalex 
Good job, thanks for sharing this wrapper!

Indeed, I had also some issue with the installation.
The reason is that the version you build from scratch (python3 setup.py build; python3 setup.py install;) is different than the one in the repository, which is different than the one that you install via 'pip3 install modbus_cli'. Same for 'pip3 install modbus_cli==0.1.4rc2': no luck to have parity option.

Can you please fix this? I think most of the people would prefer the installation via pip of the last version.

Grazie!

favalex, 2021-01-12:
Are you saying that modbus-cli has been packaged in debian? I'm not sure if there is something I should do to facilitate the coexistence of manually installations and packaged installations. I will try to find out.

I will release to pypi the latest master as soon as I have time (probably in the next few days) so at least there won't be need to install from sources to get the latest version.

dragoshenron, 2021-01-13:
AFAIK your repo isn't packaged in Debian. However, the source, the file in the /bin folder and the one you get via pip[3] should be consistent.
Maybe pull a new version like v0.1.5 might help further a user

favalex, 2021-01-17:
I've just released current master to pypi as 0.1.5.

Installing on ubuntu 18.04 with `pip install --user modbus_cli` worked for me.

I'm closing this issue. If there still are problems with installation we can open a new one.

closed 2021-01-17 by favalex