favalex/modbus-cli #11: Runtime error: zip_longest not found

ceremcem, 2021-06-09:
```
modbus -v -s 1 -b 115200 -p 0 -P n /dev/ttyUSB1 I@40008
Traceback (most recent call last):
  File "/home/ceremcem/.local/bin/modbus", line 11, in <module>
    from modbus_cli.modbus_rtu import ModbusRtu
  File "/home/ceremcem/.local/lib/python2.7/site-packages/modbus_cli/modbus_rtu.py", line 3, in <module>
    from .access import dump
  File "/home/ceremcem/.local/lib/python2.7/site-packages/modbus_cli/access.py", line 2, in <module>
    from itertools import zip_longest, groupby
ImportError: cannot import name zip_longest
```

```
sudo pip install zip_longest
Collecting zip_longest
Could not install packages due to an EnvironmentError: 404 Client Error: Not Found for url: https://pypi.org/simple/zip-longest/
```

OS: Debian Buster

favalex, 2021-06-09:
What version of python are you using? python2 is not supported. python3 should have `zip_longest`.

ceremcem, 2021-06-12:
```
$ python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import zip_longest
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'zip_longest'
```

and I can't install `zip_longest`:

```
$ sudo python3 -m pip install zip_longest
[sudo] password for ceremcem: 
Collecting zip_longest
Could not install packages due to an EnvironmentError: 404 Client Error: Not Found for url: https://pypi.org/simple/zip-longest/
```

favalex, 2021-06-12:
You have installed modbus_cli for python2.7, that won't work:

```
File "/home/ceremcem/.local/lib/python2.7/site-packages/modbus_cli/access.py"
```

python3 has `zip_longest`, but you are not testing it correctly:

```
$ python3
Python 3.6.9 (default, Jan 26 2021, 15:33:00) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from itertools import zip_longest
```

closed 2021-06-12 by favalex

ceremcem, 2021-06-12:
Ah, yes. Please consider changing install instructions as: 

```
python3 -m pip install modbus_cli
```