favalex/modbus-cli #16: Modbus RTU over TCP

mo8Zomo0, 2022-07-15:
I wonder, if it could be a cli option to use Modbus RTU over TCP. 

Sure the readme has an example for socat, but it would be more handy if this
would not be needed on the modbus-cli side.
I would like to use RS485 to ip/raw adapters and just point the modbus cli application to this adapter
and not having to add some socat rule.

favalex, 2022-07-31:
If I understand correctly how RTU over TCP works, there are devices out there that work as if they were running this step internally (referring to the example in the README):

```
remote$ socat -d -d tcp-l:54321,reuseaddr file:/dev/ttyUSB0,raw,b19200
```

I agree that is would be convenient to have a way to get rid of the `socat` step on the local side too.

Are you available to implement this improvement?

mo8Zomo0, 2022-09-19:
Hello @favalex  sadly my programming skills are not good enough. 

Yes, there are some hardware devices doing so, as well as some software.

favalex, 2022-10-09:
I won't attempt implementing this without having access to a device for testing. Here are the options I see:

1. using a publicly accessible device
2. using a simulator
3. get physical access to a device

Can you help me with any of those? Do you know of any remotely accessible device or can you suggest a simulator or device to be used for testing?