favalex/modbus-cli #24: modbus rtu slave on linux

kush930, 2023-12-18:
Hi all ,

I have **successfully installed master branch of modbus-cli on my ubuntu-20.04.3 os**, from given link- https://github.com/favalex/modbus-cli

I want to use `modbus rtu slave stack,` i am taking `my ubuntu machine as a master.`

kush@kush-Latitude-3520:~/Music/14-dec$ **modbus**

```
usage: modbus [-h] [-r REGISTERS] [-s SLAVE_ID] [-b BAUD] [-p STOP_BITS] [-P {e,o,n}] [-v] [-S] [-t TIMEOUT] [-B {le,be,mixed}] device access [access ...]
modbus: error: the following arguments are required: device, access
```

I have **energy meter,** which having `slave address  1 and register address 3027 and baudrate 9600 and device access /dev/ttyUSB0 and i want to read 16-bit data` , i am new in this, first time i am trying.

**when i am running the command,I am facing this issue-**

kush@kush-Latitude-3520:~/Music/14-dec$ `modbus -r 3027 -s 1 -b 9600 -p 1 -P n -v /dev/ttyUSB0`

```
usage: modbus [-h] [-r REGISTERS] [-s SLAVE_ID] [-b BAUD] [-p STOP_BITS]
              [-P {e,o,n}] [-v] [-S] [-t TIMEOUT] [-B {le,be,mixed}]
              device access [access ...]

modbus: error: the following arguments are required: access
```

if anyone knows where i am doing mistake, please guide me.

T&R
Kush Kumar

favalex, 2023-12-18:
Try `modbus -s 1 -b 9600 -p 1 -P n -v /dev/ttyUSB0 3027 `

`-r` is for loading a file that gives registers a name, it's optional and explained in the documentation.

kush930, 2023-12-18:
> Try `modbus -s 1 -b 9600 -p 1 -P n -v /dev/ttyUSB0 3027 `
> 
> `-r` is for loading a file that gives registers a name, it's optional and explained in the documentation.

Thank you favalex.its working now.

kush930, 2023-12-19:
Hi all,

one more issue i am facing, when i am running the command- `modbus -s 1 -b 9600 -p 1 -P n -v /dev/ttyUSB0 3027 `

facing issue- 

kush@kush-Latitude-3520:~$ **modbus -s 1 -b 9600 -p 1 -P n -v /dev/ttyUSB0 3027**

```
Parsed 0 registers definitions from 1 files
Serial port /dev/ttyUSB0. Parameters: 9600 baud, 1 stop bit(s), parity: N, timeout 5.0s.
→ < 01 03 0b d3 00 01 77 d7 >
← < 01 83 03 01 31 > 5 bytes
```
```
Traceback (most recent call last):
  File "/usr/local/bin/modbus", line 4, in <module>
    __import__('pkg_resources').run_script('modbus-cli==0.1.9', 'modbus')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 667, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1463, in run_script
    exec(code, namespace, namespace)
  File "/usr/local/lib/python3.8/dist-packages/modbus_cli-0.1.9-py3.8.egg/EGG-INFO/scripts/modbus", line 110, in <module>
    main()
  File "/usr/local/lib/python3.8/dist-packages/modbus_cli-0.1.9-py3.8.egg/EGG-INFO/scripts/modbus", line 103, in main
    connect_to_device(args).perform_accesses(parse_accesses(args.access, definitions, args.byte_order, args.silent), definitions).close()
  File "/usr/local/lib/python3.8/dist-packages/modbus_cli-0.1.9-py3.8.egg/modbus_cli/modbus_rtu.py", line 69, in perform_accesses
    access.perform(self)
  File "/usr/local/lib/python3.8/dist-packages/modbus_cli-0.1.9-py3.8.egg/modbus_cli/access.py", line 113, in perform
    self.read_registers_receive(modbus)
  File "/usr/local/lib/python3.8/dist-packages/modbus_cli-0.1.9-py3.8.egg/modbus_cli/access.py", line 146, in read_registers_receive
    words = modbus.receive(self.request)
  File "/usr/local/lib/python3.8/dist-packages/modbus_cli-0.1.9-py3.8.egg/modbus_cli/modbus_rtu.py", line 62, in receive
    return self.protocol.parse_response_adu(response, request)
  File "/usr/local/lib/python3.8/dist-packages/uModbus-1.0.4-py3.8.egg/umodbus/client/serial/rtu.py", line 190, in parse_response_adu
    function = create_function_from_response_pdu(resp_pdu, req_pdu)
  File "/usr/local/lib/python3.8/dist-packages/uModbus-1.0.4-py3.8.egg/umodbus/functions.py", line 132, in create_function_from_response_pdu
    function_code = pdu_to_function_code_or_raise_error(resp_pdu)
  File "/usr/local/lib/python3.8/dist-packages/uModbus-1.0.4-py3.8.egg/umodbus/functions.py", line 118, in pdu_to_function_code_or_raise_error
    raise error_code_to_exception_map[error_code]
umodbus.exceptions.IllegalDataValueError:  The value contained in the request data field is not an allowable value
    for the server.
```
but when i am using **number of register 2 in windows with using simply modbus, its running fine**, `Can anyone guide me how to give number of of register with using modbus command in linux.`

T&R
Kush kumar