favalex/modbus-cli #26: getting output from a temperature sensor

hboetes, 2024-04-18:
I'm trying to get data from a simple temperature sensor over the network via modbus. The device is accessible via the provided windows software, but doing it with modbus-cli would be much nicer.

The device (https://www.elen.sk/en/Products/temperature-humidity-led-displays-sensors/air-temperature-and-humidity/ths-sensor-40-poe-lan.html) has 2 registers, 0 for temperature and 10 for humidity, and I'd think I would get output like this, but I don't.

```
h.boetes@habocp3 ~ %  modbus 10.10.11.22:502 0x0             
Parsed 0 registers definitions from 1 files
h.boetes@habocp3 ~ %  modbus 10.10.11.22:502 0xa
Parsed 0 registers definitions from 1 files
h.boetes@habocp3 ~ %  modbus 10.10.11.22:502 0x10
Parsed 0 registers definitions from 1 files
```

The program uses sample template XML files like this:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<elensensorconfigurator version="2.00.00">
  <description>
    <model>TS Lan</model>
    <interface>Ethernet</interface>
    <protocol>Modbus TCP</protocol>
    <value>Temperature</value>
    <register>0</register>
    <variabletype>I = Int (2 Byte Signed):</variabletype>
    <format>0.0</format>
    <measurement>°C</measurement>
    <lowlimit>21</lowlimit>
    <highlimit>26</highlimit>
    <hysteresis>0.1</hysteresis>
  </description>
  <settings>
    <commport>No use</commport>
    <baudrate>No use</baudrate>
    <parity>No use</parity>
    <data>No use</data>
    <stopbit>No use</stopbit>
    <ipaddress>192.168.0.68</ipaddress>
    <ipport>502</ipport>
    <slaveid>1</slaveid>
  </settings>
  <lng>1051</lng>
  <info>
    <line>THS sensor with Ethernet interface.</line>
    <line></line>
    <line>Temperature [°C] - external sensor connected to Ethernet LAN.</line>
    <line></line>
    <line>MODBUS TCP protocol.</line>
    <line>Register: 0</line>   
  </info>
</elensensorconfigurator>
```

What would be right way to get the proper output?

hboetes, 2024-04-18:
Oh yes, when I don't use a hex but decimal value, I get a timeout, am I missing something?

```
h.boetes@habocp3 ~ %  modbus 10.10.11.22:502 0  
Parsed 0 registers definitions from 1 files
Traceback (most recent call last):
  File "/usr/local/bin/modbus", line 110, in <module>
    main()
  File "/usr/local/bin/modbus", line 103, in main
    connect_to_device(args).perform_accesses(parse_accesses(args.access, definitions, args.byte_order, args.silent), definitions).close()
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/modbus_cli/modbus_tcp.py", line 63, in perform_accesses
    access.read_registers_receive(self)
  File "/usr/local/lib/python3.12/site-packages/modbus_cli/access.py", line 146, in read_registers_receive
    words = modbus.receive(self.request)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/modbus_cli/modbus_tcp.py", line 29, in receive
    header = self.receive_n(6)
             ^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/modbus_cli/modbus_tcp.py", line 44, in receive_n
    data += self.connection.recv(n - len(data))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TimeoutError: timed out
```

favalex, 2024-04-18:
Hex addresses are not supported. Looks like nobody ever needed them. But it should at least report an error. What I think it's happening is that it tries to lookup the address as it it was a register name, and don't bother reporting an error if it's not defined.

The second error looks like a network problem, are you sure the device is reachable at that address? Have you tried ping or telnet?

hboetes, 2024-04-19:
Hello @favalex 

Thanks for your reply. Yes, there is a web interface running on that IP which can be used to manage the device. port 502 is open and listening, but I don't know how to get a reply from it.

```
h.boetes@habocp3 ~ %  nc 10.10.11.22 80 
sdasdsa

HTTP/1.1 400 ERROR

ERROR 400

^C
Time: 0h:00m:07s                                                                                                                                  
h.boetes@habocp3 ~ %  nc 10.10.11.22 501
Ncat: Connection refused.
h.boetes@habocp3 ~ %  nc 10.10.11.22 502
dasdasdsa
```

I think I'll set up a mirror port on the switch, so I can listen in on the traffic. See what that windows program is really sending.

hboetes, 2024-04-22:
Hello @favalex 

Here is the tcpdump capture of the communication between the sensor and the program. 

[dinges.zip](https://github.com/favalex/modbus-cli/files/15062689/dinges.zip)

I hope you can make something useful from that.