favalex/modbus-cli #37: Accessing devices with Slave-ID 0 is impossible

alphafalcon, 2024-12-02:
If I try to run 
`modbus 192.168.X.Y:502 -s 0 REGISTER `
I get a TimeoutError. Sniffing with wireshark revealed that modbus-cli was sending a slave-id of 255.
I belive that this line is responsible, as 0 is interpreted als falsy and gets replaced by 255.
https://github.com/favalex/modbus-cli/blob/dbe8939895f80d055ff4d4bc39f5eddf09a6433f/modbus_cli/modbus_tcp.py#L12

A possible fix would be adding a default of 255 to the command line argument -s and dropping the check referenced above. This should not change the behaviour, unless 0 is explicitly set on the commandline

favalex, 2024-12-14:
Hi, thanks for your report. Do you have a chance to see if https://github.com/favalex/modbus-cli/pull/38 solves your problem?

alphafalcon, 2024-12-14:
Yes, i can now send messages with unitid 0 using the linked branch, thanks!